pyrtlsdr/pyrtlsdr #153: failed initializing RtlSdr() in set_center_freq

ctx-eva, 2024-01-09:
Hello!
Do Blog v4 supported with pyrtlsdr. I suffered the problem Could not set center_freq with DEFAULT_FC when initializing RtlSdr() on Ubuntu 20.04 with python 3.8.10.

The device is as follow:
Found 1 device(s):
  0:  RTLSDRBlog, Blog V4, SN: 00000001

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R828D tuner
RTL-SDR Blog V4 Detected

Python 3.8.10 (default, Nov 22 2023, 10:22:35) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from rtlsdr import *
>>> sdr = RtlSdr()
Found Rafael Micro R828D tuner
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/hym/.local/lib/python3.8/site-packages/rtlsdr/rtlsdr.py", line 133, in __init__
    self.open(device_index, test_mode_enabled, serial_number)
  File "/home/hym/.local/lib/python3.8/site-packages/rtlsdr/rtlsdr.py", line 184, in open
    self.init_device_values()
  File "/home/hym/.local/lib/python3.8/site-packages/rtlsdr/rtlsdr.py", line 198, in init_device_values
    self.set_center_freq(float(self.DEFAULT_FC))
  File "/home/hym/.local/lib/python3.8/site-packages/rtlsdr/rtlsdr.py", line 218, in set_center_freq
    raise LibUSBError(result, 'Could not set center_freq to %d Hz' % (freq))
rtlsdr.rtlsdr.LibUSBError: <LIBUSB_ERROR_IO (-1): Input/output error> "Could not set center_freq to 80000000 Hz"

tcasaniv, 2024-01-22:
The same thing happens to me in Windows 11 and virtualenv.

I also think this is because RTL-SDR V4 is not yet supported.

```pwsh
PS C:\Users\tcasaniv> py
Python 3.12.1 (tags/v3.12.1:2305ca5, Dec  7 2023, 22:03:25) [MSC v.1937 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from rtlsdr import RtlSdr
>>> sdr = RtlSdr()
Found Rafael Micro R828D tuner
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\tcasaniv\AppData\Local\Programs\Python\Python312\Lib\site-packages\rtlsdr\rtlsdr.py", line 133, in __init__
    self.open(device_index, test_mode_enabled, serial_number)
  File "C:\Users\tcasaniv\AppData\Local\Programs\Python\Python312\Lib\site-packages\rtlsdr\rtlsdr.py", line 184, in open
    self.init_device_values()
  File "C:\Users\tcasaniv\AppData\Local\Programs\Python\Python312\Lib\site-packages\rtlsdr\rtlsdr.py", line 198, in init_device_values
    self.set_center_freq(self.DEFAULT_FC)
  File "C:\Users\tcasaniv\AppData\Local\Programs\Python\Python312\Lib\site-packages\rtlsdr\rtlsdr.py", line 218, in set_center_freq
    raise LibUSBError(result, 'Could not set center_freq to %d Hz' % (freq))
rtlsdr.rtlsdr.LibUSBError: <LIBUSB_ERROR_IO (-1): Input/output error> "Could not set center_freq to 80000000 Hz"
>>>
```

nocarryr, 2024-02-02:
@tcasaniv I believe you're correct.  IIRC, they added support for the V4 somewhat recently and I don't believe that version is what's compiled into the sister project [pyrtlsdrlib](https://github.com/pyrtlsdr/pyrtlsdrlib).

Until another build/release can be triggered, you may want to try compiling [librtlsdr](https://github.com/librtlsdr/librtlsdr) manually

JJones4452, 2024-04-27:
Using on Windows 10 and Python 3.12.1.

I had this error and seemed to fix it by removing the pyrtlsdrlib package and adding a folder containing the drivers from the RTL-SDR blog [quick start guide](https://www.rtl-sdr.com/V4/) to my Windows path (though I haven't tested if this part is required). 

It seems that if the pyrtlsdrlib package is installed, it defaults to using the dll that's packaged within it which doesn't seem compatible with the V4. Doing this allows me to run the getting started code.

Hope this helps.

nocarryr, 2024-04-27:
@JJones4452 
Support has been added for this in [pyrtlsdrlib v0.0.3](https://github.com/pyrtlsdr/pyrtlsdrlib/releases/tag/0.0.3)

So if you do a pip upgrade (`pip install -U pyrtlsdrlib`) that should work without having to mess with dll's.

You are correct though, if it's installed, the included library files will be the first preference when pyrtlsdr loads (before searching any platform-specific locations)