pyrtlsdr/pyrtlsdr #158: RTL-SDR V4 - "Could not set center_freq to 80000000 Hz"

PinkiMan, 2024-03-24:
Hello, I am starting with RTL-SDR V4 and I can't run class init with this device.

```
> from rtlsdr import RtlSdr
> 
> sdr = RtlSdr()
```

And this is the output it gives:

> rtlsdr.rtlsdr.LibUSBError: <LIBUSB_ERROR_IO (-1): Input/output error> "Could not set center_freq to 80000000 Hz"
Found Rafael Micro R828D tuner

Is there some way to make it working?

stevennyman, 2024-03-30:
I am also experiencing this issue with the RTL-SDR V4, both on Windows 10 and on Linux (Ubuntu 22.04).

stevennyman, 2024-03-31:
It appears that `pyrtlsdr` bundles a slightly outdated version of `librtlsdr`. Removing the bundled version of the library and manually downloading the [newest release](https://github.com/librtlsdr/librtlsdr/releases) from last week, whitelisting it in my antivirus software (it's a new file so it was unrecognized), and adding it to my PATH resolved the issue.

PinkiMan, 2024-03-31:
Ohhh, niceeee, so I need to download `librtlsdr`. I am on windows 10, so what exactly from librtlsdr I need to add to PATH?

stevennyman, 2024-03-31:
The bundled version of the library is installed when you install `pyrtlsdr[lib]`. You can uninstall it using `pip uninstall pyrtlsdrlib`. As for adding `librtlsdr` to my path, I extracted all the files to their own directory and then just added that directory to my PATH. It's possible that you only need to add the `librtlsdr.dll` file to PATH but I haven't tested it.

PinkiMan, 2024-03-31:
Thanks a lot, uninstalling worked, but now adding it to PATH is problem. I downloaded: v0.9 `[rtlsdr-bin-w64_static.zip]`, unzip it, than added it to PATH in system variables : `C:\Users\...\rtlsdr-bin-w64_static`, but it is still not working. I found that it is looking for at least one from these files:

```
> driver_files += ['librtlsdr.so', 'rtlsdr/librtlsdr.so']
> driver_files += ['rtlsdr.dll', 'librtlsdr.so', 'librtlsdr.dylib']
> driver_files += ['..//rtlsdr.dll', '..//librtlsdr.so']
> driver_files += ['rtlsdr//rtlsdr.dll', 'rtlsdr//librtlsdr.so'] 
```

Really thanks for your help

stevennyman, 2024-03-31:
Try logging out and logging back in to your computer.

PinkiMan, 2024-03-31:
Damn, thanks a lot, that worked. Finally. You are legend

nocarryr, 2024-03-31:
@stevennyman @PinkiMan 
Just pushed a new release of [pyrtlsdrlib](https://github.com/pyrtlsdr/pyrtlsdrlib/releases/tag/0.0.3) with librtlsdr v0.9.0.

If you want to give it a shot, you should be able to update via pip:

```bash
pip install -U pyrtlsdrlib
```

And you wouldn't have to undo all of the PATH manipulation from above since pyrtlsdr looks for the packaged libs before searching elsewhere.

Let me know if that works for you and I'll close this issue

stevennyman, 2024-03-31:
Yes this works for me, thanks for updating it!

closed 2024-04-01 by nocarryr