pyrtlsdr/pyrtlsdr #159: RTL-SDR V4 on Mac: librtlsdr are not in PATH error

PinkiMan, 2024-04-03:
Hello, I working on Mac-Sonoma 14.4.1 and just moved from Win10, where it is working!!!  
Same code, same installation. RTL-SDR working in SDR++.

When running init of `RtlSdr` lib it raise message and error: 

```

> Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib, 0x0006): Library not loaded: /opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib\n  Referenced from: <27CD318A-15CD-3759-8DBD-F9EAFBB7CF68> /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib\n  Reason: tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file)")
> Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib, 0x0006): Library not loaded: /opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib\n  Referenced from: <27CD318A-15CD-3759-8DBD-F9EAFBB7CF68> /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib\n  Reason: tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file)")
> Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib, 0x0006): Library not loaded: /opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib\n  Referenced from: <27CD318A-15CD-3759-8DBD-F9EAFBB7CF68> /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib\n  Reason: tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file)")
> Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib, 0x0006): tried: '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))")
> Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib, 0x0006): tried: '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))")
> Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib, 0x0006): tried: '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))")
> Traceback (most recent call last):
>   File "/Users/..../sdr.py", line 2, in <module>
>     from rtlsdr import *
>   File "/Users/...../Python/3.9/lib/python/site-packages/rtlsdr/__init__.py", line 56, in <module>
>     from .librtlsdr import librtlsdr
>   File "/Users/...../Python/3.9/lib/python/site-packages/rtlsdr/librtlsdr.py", line 62, in <module>
>     librtlsdr = load_librtlsdr()
>   File "/Users/...../Python/3.9/lib/python/site-packages/rtlsdr/librtlsdr.py", line 57, in load_librtlsdr
>     raise ImportError('Error loading librtlsdr. Make sure librtlsdr '\
> ImportError: Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path

```

Standard installation:
```
> pip install pyrtlsdr
> pip install -U pyrtlsdr
```

Also tried downloading `librtlsdr v0.9` for Mac and manually adding to PATH, but same error.

Can anyone help please?

nocarryr, 2024-04-03:
MacOS has different env variables for dynamic library paths.  IIRC, it's "DYLD_LIBRARY_PATH" but I'm not 100% sure.  You could mess around with that, but have you tried it with the "pyrtlsdrlib" helper package?  It might be easier.

Can be installed manually with:
```bash
pip install pyrtlsdrlib
```

Or as a "pip extra":
```bash
pip install 'pyrtlsdr[lib]'
```

PinkiMan, 2024-04-03:
I tried the 'pip extra' "pyrtlsdrlib" helper package right now, but it did not worked, same error sadly. 

What could I do with different MacOS dynamic library paths? Should I edit some files or something?

Thanks for your help

nocarryr, 2024-04-03:
I'm curious about why the helper package didn't work.  It shouldn't have given the same error if it's installed.  What happens if you try this in your python interpreter?

```python
import pyrtlsdrlib
print(pyrtlsdrlib.get_library_files())
```

PinkiMan, 2024-04-03:
It returns this

> `[PosixPath('/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib'), PosixPath('/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib'), PosixPath('/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib'), PosixPath('/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib'), PosixPath('/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib'), PosixPath('/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib')]
> `

PinkiMan, 2024-04-03:
Sounds it is getting to some point. Reponse:

`Traceback (most recent call last):
  File "/Users/pinki/PycharmProjects/My_Utils_V1/.archive/test.py", line 2, in <module>
    dll = pyrtlsdrlib.load_library()
AttributeError: module 'pyrtlsdrlib' has no attribute 'load_library'`

nocarryr, 2024-04-03:
Hmmm... I'm wondering if the packaging got messed up somehow on OSX with the latest release.  I'll have to check on a mac a bit later today.

Are you installing things 'system-wide'  or using a virtual environment?  I would recommend the latter as system-wide packages are rarely a good idea and testing for both of these projects is always done in a virtualenv.

You might try uninstalling both `pyrtlsdr` and `pyrtlsdrlib` from the global site packages then creating and using a venv:

```bash
pip uninstall pyrtlsdrlib pyrtlsdr

# create a virtual environment in the current working directory named ".venv"
python -m venv .venv

# activate it
source .venv/bin/activate
pip install pyrtlsdr[lib]
```

The environment can then be deactivated by
```bash
deactivate
```

I'll let you know if I found any issues on my macbook later on

PinkiMan, 2024-04-03:
Ok, thanks I will try it.

Let me know if you found out if it works. It could be my local problem

nocarryr, 2024-04-03:
Whoops!  I just realized I gave you the wrong test code.  It should be:

```python
import pyrtlsdrlib
dll = pyrtlsdrlib.load_librtlsdr()
print(repr(dll))
```

Works fine on my mac

PinkiMan, 2024-04-03:
Damn, my bad I should had notice it.

```
Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib, 0x0006): Library not loaded: /opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib\n  Referenced from: <27CD318A-15CD-3759-8DBD-F9EAFBB7CF68> /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.dylib\n  Reason: tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file)")
Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib, 0x0006): Library not loaded: /opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib\n  Referenced from: <27CD318A-15CD-3759-8DBD-F9EAFBB7CF68> /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.0.9git.dylib\n  Reason: tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file)")
Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib, 0x0006): Library not loaded: /opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib\n  Referenced from: <27CD318A-15CD-3759-8DBD-F9EAFBB7CF68> /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/custom_build/librtlsdr.dylib\n  Reason: tried: '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/opt/homebrew/opt/libusb/lib/libusb-1.0.0.dylib' (no such file)")
Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib, 0x0006): tried: '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))")
Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib, 0x0006): tried: '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.0.8git.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))")
Could not load /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib. Exception: OSError("dlopen(/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib, 0x0006): tried: '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64')), '/System/Volumes/Preboot/Cryptexes/OS/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/pyrtlsdrlib/lib/librtlsdr.dylib' (mach-o file, but is an incompatible architecture (have 'x86_64', need 'arm64e' or 'arm64'))")
None

```

PinkiMan, 2024-04-03:
I just tried the virtual environment, that you said to try. It returns same error:

```
raise ImportError('Error loading librtlsdr. Make sure librtlsdr '\
ImportError: Error loading librtlsdr. Make sure librtlsdr (and all of its dependencies) are in your path
```

nocarryr, 2024-04-03:
I tested it successfully on an m1 macbook pro.

I'd suggest:
- Undoing all of the stuff you did in the first post (custom install of librtlsdr and whatever you did to the "PATH").
- Uninstall `pyrtlsdr` and `pyrtlsdrlib` from your system-wide packages (`pip uninstall <...>` from *outside* of any virtual environment
- Make sure that `import rtlsdr` **and** `import pyrtlsdrlib` both throw an `ImportError` just to be sure

You want to be sure it's not trying to find any of those library files.

Then hop back into your virtual environment and ensure you can `import pyrtlsdrlib` and check that `print(pyrtlsdrlib.__file__)` shows a path inside of the venv.
If all's good, try the `dll = pyrtlsdrlib.load_librtlsdr()` thing again.

It's possible that it needs `libusb` (that's in one of your error logs).  That *should* be ok to install through hombrew (`brew install libusb`).  If that ends up working, I'll add some documentation somewhere to avoid issues in the future

PinkiMan, 2024-04-03:
Yesssss, finally working, big thanks, you are legend. 

`Homebrew` installed and than after `linusb` installed it all started working

closed 2024-04-04 by nocarryr