pyrtlsdr/pyrtlsdr #169: Issue with Importing pyrtlsdr in WSL (Ubuntu) Virtual Environment

ceauce, 2024-12-02:
hi there

Environment:
	•	OS: Windows 10/11 with WSL (Ubuntu)
	•	Python version: 3.12.3
	•	pyrtlsdr version: 0.3.0
	•	Other dependencies: scipy, numpy, matplotlib

Description:
I am facing an issue when trying to use the pyrtlsdr library in a Python virtual environment on WSL (Ubuntu). Despite installing the library and its dependencies, I am unable to import pyrtlsdr in my Python script. I have followed the steps below to troubleshoot the problem, but the issue persists.

Steps Taken:
	1.	Created a new Python virtual environment (python3.10 -m venv myenv).
	2.	Installed the necessary libraries using pip install pyrtlsdr matplotlib numpy scipy.
	3.	Verified the installation with pip show pyrtlsdr (it is correctly installed in /home/paul/myenv/lib/python3.12/site-packages).
	4.	Verified the Python version and ensured it’s pointing to the virtual environment: python3 --version shows Python 3.12.3.
	5. 
	thanks

georgeh1ll, 2024-12-17:
Hi, I managed to fix it on windows using 
`pip install setuptools` 
(possibly with sudo in linux case).

eabase, 2025-02-20:
@ceauce 
Did you get this to work?